PaulStoffregen/OctoWS2811 #6: PlatformIO directory problem

amittel, 2016-01-24:
Hi!
I installed your lib via
``platformio lib install 69``

But I got a compile error not finding the lib.

The problem was, that the files were in the wrong directory.
In Windows7 they were in
``%USER%\.platformio\lib\OctoWS2811_ID69\OctoWS2811``
but they should be in
``%USER%\.platformio\lib\OctoWS2811_ID69``

Copied ``OctoWS2811.cpp`` and ``OctoWS2811.h`` into ``OctoWS2811_ID69`` and now it works as expected.

PaulStoffregen, 2016-01-24:
@ivankravets - Is this something I should fix here, or something needed inside Platform IO?

ivankravets, 2016-01-24:
@PaulStoffregen please merge #7 

P.S: What do you think about [PlatformIO IDE for Atom](http://docs.platformio.org/en/latest/ide/atom.html) ?

PaulStoffregen, 2016-01-24:
done

PaulStoffregen, 2016-01-24:
> P.S: What do you think about PlatformIO IDE for Atom ?

This is the first I've seen it.  Too busy with urgent projects to try it right now... maybe in a couple months?

closed 2016-01-24 by PaulStoffregen

ivankravets, 2016-01-24:
@amittel please `platformio lib update`. See http://platformio.org/#!/lib/show/69/OctoWS2811

ivankravets, 2016-01-24:
> This is the first I've seen it. 

I've released first beta a few hours ago :blush: 

> Too busy with urgent projects to try it right now... maybe in a couple months?

Sure! I've just asked to "not try", but just take a look on the features. I think that it will be good alternative to users who want more than Arduino IDE. However, I don't know how Arduino community will react on it :(

PaulStoffregen, 2016-01-24:
I get quite a few requests from people who want more powerful features than Arduino's simplistic IDE.  Maybe this will be a good path?

ivankravets, 2016-01-24:
> Maybe this will be a good path?

I hope that YES. Some users say that the first PlatformIO IDE beta is "better 1000x" than current Arduino IDE. However, I'm not so optimistic in this comparison. Firstly, need to implement Library Manager and built-int Serial Monitor. Yes, PlatformIO DE contains built-in Terminal and user can use PlatformIO CLI within it, but they like GUI.

PaulStoffregen, 2016-01-24:
Maybe start a thread on the forum, inviting people to give it a try?

https://forum.pjrc.com/forums/5-General-Discussion

amittel, 2016-01-25:
Did the update and now everything looks good.

ivankravets, 2016-03-08:
> Maybe start a thread on the forum, inviting people to give it a try?

@PaulStoffregen, could you pin this thread?

https://forum.pjrc.com/threads/33388-PlatformIO-IDE-The-next-generation-IDE-for-IoT-development?p=98647

Thanks!

commit c7e3f337 (2016-01-24): Merge pull request #7 from ivankravets/patch-2

Use tagged source code from repo // Resolve #6
commit cb873621 (2016-01-24): Use tagged source code from repo // Resolve #6