PaulStoffregen/OctoWS2811 #12: question

jackthemaker, 2016-10-31:
I am seeing people fleeing from ws2811 to sk6812

does this library works with that chip , Adafruit says that they are technically the same thing?

Thx

ColinHarrington, 2016-10-31:
I haven't used any of the sk6812 LEDs, but [the datasheet](http://www.szledcolor.com/download/SK6812%20LED.pdf) looks promising. 

Tim seems to think that "The SK6812 is a direct clone of the WS2812B" - [source](https://cpldcpu.wordpress.com/2016/03/09/the-sk6812-another-intelligent-rgb-led/)

There may be some slight timing differences.  Please report any issues.

PaulStoffregen, 2016-11-12:
I recently purchased some SK6812 LEDs.  Haven't had time yet to do anything with them, but there's a fresh reel of them sitting on my desk, in a big pile of other stuff I haven't had time to look at either...