PaulStoffregen/OctoWS2811 #17: Rainbow example doesn't properly set the first pixel with Arduino 1.8.2/Teensyduino 1.36

ssilverman, 2017-05-10:
Arduino 1.6.12 and Teensyduino 1.31 is the latest combination that works for this demo for me.

PaulStoffregen, 2017-05-10:
Yup, looks like a bug was recently added which affects some (but not all) LED strips.

I will look into this within the next few weeks.

SpencerGreene, 2017-05-25:
+1 on this.  happy to look into it if you have any guesses / hints where it might be?

davydka, 2017-05-25:
Just ran into this issue as well!

PaulStoffregen, 2017-06-15:
I've uploaded a new version with OctoWS2811 fixes.  Please give it a try and let me know if it solves this problem on your LED strips.

https://forum.pjrc.com/threads/44904-Teensyduino-1-37-Beta-3

ssilverman, 2017-08-15:
It looks like that fixes it. Can a second party confirm this?

PaulStoffregen, 2017-08-25:
Yes, several people have confirmed on the forum.

I'm closing this old issue.

closed 2017-08-25 by PaulStoffregen