PaulStoffregen/OctoWS2811 #18: Glediator example problem

magveda, 2017-05-24:
Hello,

I am using [Glediator](https://github.com/PaulStoffregen/OctoWS2811/tree/master/examples/Glediator) example and the LEDs on even lines are shifted by 2 pixels:

[image]

The text should be "olab.lt"

There is no problem when I use this code:

```
#include "FastLED.h"
#define NUM_LEDS 60*18
const int dataline = 2;

CRGB leds[NUM_LEDS];
void setup() {
        Serial.begin(115200); 
     LEDS.addLeds<WS2812B, dataline>(leds, NUM_LEDS);
}
int serialGlediator() {
  while (!Serial.available()) {}
  return Serial.read();
}
void loop() {
   while (serialGlediator() != 1) {} 

   for (int i=0; i < NUM_LEDS; i++) {
     leds[i].r = serialGlediator();
     leds[i].g = serialGlediator();
     leds[i].b = serialGlediator();
   }
     FastSPI_LED.show();
}
```
But this is working only for leds on data pin 2. I have 2 panels on 2 pins - 2 and 14.

How can I resolve this issue?

PaulStoffregen, 2017-05-24:
Of course it only uses pin 2.  You're not using OctoWS2811 in any way in this code!

magveda, 2017-06-03:
Hey Paul. Maybe my question is not clear enough. I am using 2 code samples. One using OctoWS2811 and other using FastLED library. The one with OctoWS2811 is shifting pixels. And the one with FastLED is working good.

I updated FastLED to use pins 2 and 14 and everything is good now. But no success with OctoWS2811.