PaulStoffregen/OctoWS2811 #22: Reduce/remove LED_HEIGHT multiple of 8 constrain

fabianmoronzirfas, 2017-10-26:
Hi. We are trying to use your library to drive a lot of WS2812B LEDs. Now we are running into performance issues because of the length and height of a section of our display.

We are using on the teensy your firmware `VideoDisplay.ino` and drive it from openFrameworks with a port of the `movie2serial.pde` code.

Is there any method to reduce the **"LED_HEIGHT must be a multiple of 8 constrain"** and maybe get it down to 3 strips in the height?  

Any help or hint would be highly appreciated.

PaulStoffregen, 2017-10-26:
This can not be changed.  But you can simply use only 3 of the 8 channels and "waste" the other 5.

closed 2017-10-26 by PaulStoffregen

fabianmoronzirfas, 2017-10-26:
Thanks for your fast reply. Yes we are doing that, but still get a huge lag. We have on one teensy 3.2  a LED_WIDTH of around 250 LEDs that means when multiplied with 8 its 2000 pixels it tries to drive. 

I took a look around and found that the FASTLED library has a [Octo WS2811](https://github.com/FastLED/FastLED/blob/master/examples/Multiple/OctoWS2811Demo/OctoWS2811Demo.ino) adaptor. Any chance this might help? And I actually don't know how to hook in the FASTLED lib instead of yours

PaulStoffregen, 2017-10-26:
What is "a huge lag"?

OctoWS2811 should update 2000 LEDs in 7.5 ms.

fabianmoronzirfas, 2017-10-26:
Okay. Then it might be a problem with my openFrameworks code. When I try to run the a single board with 250LEDs in the width it nearly stops updating the pixels. I will give it a try with your original processing code.

PaulStoffregen, 2017-10-26:
If you need help, post this on the forum.  Try to compose a message with all materials needed to reproduce the problem, including step-by-step instructions for people (like me) who've never used openframeworks.

These github issues are for reporting actual bugs.  Please, move this to the forum until it's confirmed to really be a bug in OctoWS2811.

Also, if you are using Teensy 3.5, please update to the latest code from github.  A bug affecting only T3.5 was recently fixed.

fabianmoronzirfas, 2017-10-26:
Okay. Just a last remark. I used your original processing example and it runs smooth. It is a problem on my OF code. Sorry for the fuzz and thanks for your help.