PaulStoffregen/OctoWS2811 #24: WS2813 frameset delay

xseignard, 2017-11-29:
Hello,

Maybe I'm wrong but the commit by @alex-Arc (https://github.com/PaulStoffregen/OctoWS2811/pull/9) disappeared from the master branch.

Is that normal?

Regards,

Xavier

xseignard, 2017-11-29:
Ok, I just saw you replaced the frameset delay param by hardcoding the biggest value (300μs) needed by WS2811 (50μs) and WS2813 (300μs) here: https://github.com/PaulStoffregen/OctoWS2811/blob/master/OctoWS2811.cpp#L247

Sorry for the noise, and again, thanks to all for your quality work 👍

closed 2017-11-29 by xseignard

PaulStoffregen, 2017-11-29:
About a year ago, Worldsemi made a new chip which requires 300 us, but they called it WS2812B.  There's no practical way for anyone to be able to tell which WS2812B they actually have, so I just removed the old 50 us time.

ColinHarrington, 2017-11-29:
I think that hardcoding the larger 300µs isn't the right way to go.  Maybe making 300µs the default makes sense, but you should be able to configure it back to the 50µs window.  At 60 updates per second, that is 18,000µs (18 milliseconds) which is an additional 15ms vs the original 50µs window.  The timing differences aren't insignificant in high-framerate situations.

PaulStoffregen, 2017-11-30:
Remember, this is not a fixed delay.  It only delays if you call show() before the prior frame + 300 us is complete.  If driving 60 Hz video onto LED is your project, presumably your code will control the timing by calling show() exactly 60 times per second.  If you have less than 4360 LEDs (545 per pin), the LEDs will fully update and more than 300 us will have passed by the time you call show again.  There will be no delay inside the library.  You'll get the full CPU time to do whatever work you need to get the next frame ready.

ColinHarrington, 2017-12-01:
Yea the reset time isn't a `delay()` which blocks execution, but is a delay for when you can start writing pixel data out again.  If you do end up calling `show()` before the window is elapsed, it will block until the interrupt clears `update_in_progress` and the reset window has elapsed.  

The 545 pixels per pin in the absolute max you can squeeze into that time with (800khz data rate) and get 60 frames per second.   With a reset time 50µs that is 553pixels.  

It really starts to matter when you are pushing high framerates for temporal dithering. I'd still prefer that window to be configurable.  BTW the SK6812s(WS2812b clone) have a reset time of 80µs, so my vote is still to allow some form of configuration of the reset time.