PaulStoffregen/OctoWS2811 #31: Teensy 4.0

lovettchris, 2019-10-04:
Can you add support for Teensy 4.0 ?

When I try and compile your BasicTest example for Teensy 4.0 I get this error:

```
OctoWS2811.cpp:85:2: error: 'GPIOD_PCOR' was not declared in this scope
```

BarakChamo, 2019-10-05:
I experienced the same issue, the `Teensy 4` board definitions don't contain the 
`// General-Purpose Input/Output (GPIO)` in `kinetis.h` or anywhere else, it's only present under `teensy3`

egapevets2, 2020-01-14:
I would also appreciate Teensy 4.0 support

GvBon, 2020-03-10:
There is an update for teensy 4 three days ago by Paul. 
I guess you can set the pins in pinlist? Clarity anyone?

PaulStoffregen, 2020-03-10:
Yes

GvBon, 2020-03-10:
ah nice! 
the error is a problem on my side then? (very well possible)

PaulStoffregen, 2020-03-10:
I just compiled this without any errors.

```
const int ledsPerStrip = 240;

DMAMEM int displayMemory[ledsPerStrip*6];
int drawingMemory[ledsPerStrip*6];

const int config = WS2811_GRB | WS2811_800kHz;

uint8_t pins[5] = {2, 3, 4, 5, 6};

OctoWS2811 leds(ledsPerStrip, displayMemory, drawingMemory, config, 5, pins);
```

In File > Prefs, turn on verbose output.  Then look for info about the folder Arduino is actually using for the library.  Maybe it's not using the copy you've installed?

Or wait a few days... soon we'll being beta testing 1.52.  Then you can just run the installer to get the latest version of the library.

GvBon, 2020-03-10:
Thanks! Cheers to Teensy 4.0!

GvBon, 2020-03-11:
Hmm, tried that. Library now is the right one.
But still running into errors:
`C:\Users\gijsv\OneDrive\Documents\Arduino\libraries\OctoWS2811-master\OctoWS2811_imxrt.cpp: In member function 'void OctoWS2811::begin()':

C:\Users\gijsv\OneDrive\Documents\Arduino\libraries\OctoWS2811-master\OctoWS2811_imxrt.cpp:149:17: error: 'XBARA_CTRL_STS1' was not declared in this scope

  XBARA1_CTRL0 = XBARA_CTRL_STS1 | XBARA_CTRL_EDGE1(3) | XBARA_CTRL_DEN1 |

                 ^

C:\Users\gijsv\OneDrive\Documents\Arduino\libraries\OctoWS2811-master\OctoWS2811_imxrt.cpp:149:53: error: 'XBARA_CTRL_EDGE1' was not declared in this scope

  XBARA1_CTRL0 = XBARA_CTRL_STS1 | XBARA_CTRL_EDGE1(3) | XBARA_CTRL_DEN1 |

                                                     ^

C:\Users\gijsv\OneDrive\Documents\Arduino\libraries\OctoWS2811-master\OctoWS2811_imxrt.cpp:149:57: error: 'XBARA_CTRL_DEN1' was not declared in this scope

  XBARA1_CTRL0 = XBARA_CTRL_STS1 | XBARA_CTRL_EDGE1(3) | XBARA_CTRL_DEN1 |

                                                         ^

C:\Users\gijsv\OneDrive\Documents\Arduino\libraries\OctoWS2811-master\OctoWS2811_imxrt.cpp:150:3: error: 'XBARA_CTRL_STS0' was not declared in this scope

   XBARA_CTRL_STS0 | XBARA_CTRL_EDGE0(3) | XBARA_CTRL_DEN0;

   ^

C:\Users\gijsv\OneDrive\Documents\Arduino\libraries\OctoWS2811-master\OctoWS2811_imxrt.cpp:150:39: error: 'XBARA_CTRL_EDGE0' was not declared in this scope

   XBARA_CTRL_STS0 | XBARA_CTRL_EDGE0(3) | XBARA_CTRL_DEN0;

                                       ^

C:\Users\gijsv\OneDrive\Documents\Arduino\libraries\OctoWS2811-master\OctoWS2811_imxrt.cpp:150:43: error: 'XBARA_CTRL_DEN0' was not declared in this scope

   XBARA_CTRL_STS0 | XBARA_CTRL_EDGE0(3) | XBARA_CTRL_DEN0;

                                           ^

C:\Users\gijsv\OneDrive\Documents\Arduino\libraries\OctoWS2811-master\OctoWS2811_imxrt.cpp: In member function 'void OctoWS2811::show()':

C:\Users\gijsv\OneDrive\Documents\Arduino\libraries\OctoWS2811-master\OctoWS2811_imxrt.cpp:259:18: error: 'XBARA_CTRL_STS1' was not declared in this scope

  XBARA1_CTRL0 |= XBARA_CTRL_STS1 | XBARA_CTRL_STS0;

                  ^

C:\Users\gijsv\OneDrive\Documents\Arduino\libraries\OctoWS2811-master\OctoWS2811_imxrt.cpp:259:36: error: 'XBARA_CTRL_STS0' was not declared in this scope

  XBARA1_CTRL0 |= XBARA_CTRL_STS1 | XBARA_CTRL_STS0;

                                    ^
`

PaulStoffregen, 2020-03-11:
Oh, yeah, the new code depends on updates that were recently made to the core library.

Probably simplest to just wait for a 1.52-beta1 installer that puts every file in the right place.

PaulStoffregen, 2020-03-11:
1.52-beta1 installers are now available.

https://forum.pjrc.com/threads/59996-Teensyduino-1-52-Beta-1

GvBon, 2020-03-11:
Testing right now with a teensy4 and 16 pins {1, 0, 24, 25, 19, 18, 14, 15, 17, 16, 22, 23, 20, 21, 26, 27}  connected through 74hct245 and 100Ohm resistor.
8 strips of 300 led's (ws2815b, 1 strip per pin) run beautifully at about 100fps on one teensy 4.
More than 350 per strip seems to be running more into chaos. Seems to be a timing issue. I have been playing around with several delays but 
more than 8 strips do not run (the first 8 do, after 8 they do not, possibly because this is OCTO -> 8 (?)).

PaulStoffregen, 2020-03-19:
> More than 8 strips do not run

Please post on the forum.  Show the complete program you're using.

> possibly because this is OCTO -> 8 (?)).

Internally the Teensy 4.0 code is using 128 bits (not only 8), so it can support any combination of all digital pins.

closed 2020-03-19 by lovettchris