PaulStoffregen/OctoWS2811 #32: ERROR TO COMPILE FOR ARDUINO MEGA

NACHOAMERI, 2020-03-26:
Hi, someone can help me, i want to use this code to control by jinx led strip ws2811... but it's throws me "error to compile for arduino  mega"

Thanks!!

PaulStoffregen, 2020-03-26:
Can you show a screenshot of the actual error?

It is supposed to say: "Sorry, OctoWS2811 only works on 32 bit Teensy boards.  AVR isn't supported."

NACHOAMERI, 2020-03-27:
Thats a the error!

Arduino:1.8.12 (Windows Store 1.8.33.0) (Windows 10), Tarjeta:"Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from C:\Users\Nacho Ameri\Desktop\OctoWS2811-b0df1189c8156115a8951b72c56d59ab7d710b76\examples\Glediator\Glediator.ino:9:0:

C:\Users\Nacho Ameri\Documents\Arduino\libraries\OctoWS2811-b0df1189c8156115a8951b72c56d59ab7d710b76/OctoWS2811.h:28:2: error: #error "Sorry, OctoWS2811 only works on 32 bit Teensy boards.  AVR isn't supported."

 #error "Sorry, OctoWS2811 only works on 32 bit Teensy boards.  AVR isn't supported."

  ^~~~~

In file included from C:\Users\Nacho Ameri\Desktop\OctoWS2811-b0df1189c8156115a8951b72c56d59ab7d710b76\examples\Glediator\Glediator.ino:9:0:

C:\Users\Nacho Ameri\Documents\Arduino\libraries\OctoWS2811-b0df1189c8156115a8951b72c56d59ab7d710b76/OctoWS2811.h:32:10: fatal error: DMAChannel.h: No such file or directory

 #include "DMAChannel.h"

          ^~~~~~~~~~~~~~

compilation terminated.

exit status 1
Error compilando para la tarjeta Arduino Mega or Mega 2560.

Este informe podría contener más información con
"Mostrar salida detallada durante la compilación"
opción habilitada en Archivo -> Preferencias.

PaulStoffregen, 2020-03-27:
Arduino Mega uses AVR.  Hopefully you can understand this error?

closed 2020-03-27 by PaulStoffregen